RoadkillUK/Motor-on-a-Roller-Blind-for-ESPHOME #3: Disable status LED on board

davefrooney, 2021-02-23:
Thanks for your great code. I have setup 4 blinds with this and they are great. 

I am wondering is it possible to disable the blue led on the D1 mini? It can be very bright in a dark room. I looked on esphome and it looks like the LED is tied to the same pins as your code. I don't know enough about this but maybe it is possible to do?

RoadkillUK, 2021-02-23:
I've not looked into it at all, sorry, how about a bit of electrical tape over the LED :)

cmccambridge, 2021-03-06:
Hey @davefrooney - yes, this is definitely possible. If you wire your motor coil to a different pair of pins that does **not** include `D4`, then you'll avoid turning on and off the D1 mini's LED. I would suggest any of `D5`, `D6`, or `D7`. You'll just need to update `D4` to the corresponding other value in the settings in your `.yaml` file. 

You can use any compatible (see below) GPIO pins for driving the motor, there's nothing **requiring** you to use `D1` through `D4` other than that they're the defaults in the source code here - @RoadkillUK made them easy to swap out by changing only the settings in the `.yaml` file 😄 

(Thanks for this work @RoadkillUK ! I've learned a lot of ESPHome tricks from reading your code 😄)

Here's a page that I often refer back to with details on the pinouts and what other functions are shared by each. See the section "Best Pins to use - ESP8266" for a handy chart. https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

RoadkillUK, 2021-03-06:
Thanks @cmccambridge, I only use these pins as this sketch is meant to be a direct replacement for the original [MOARB](https://www.thingiverse.com/thing:2392856).

cmccambridge, 2021-03-06:
👍 yep makes sense! Just noting for @davefrooney 's benefit that switching to a different pin is an option without changing functionality, and will avoid turning on the LED since the LED is physically wired to `D4`

RoadkillUK, 2021-03-26:
Looks like the fix for this is not to use pin `D4`

closed 2021-03-26 by RoadkillUK