RoadkillUK/Motor-on-a-Roller-Blind-for-ESPHOME #7: after the esp8266 is powered off

qqxioud, 2022-10-07:
Hello, do you need to recalibrate after the esp8266 is powered off?

RoadkillUK, 2022-10-07:
Doesn't need calibrating after powering off, unless it was actually moving at the time.

closed 2022-10-07 by RoadkillUK

qqxioud, 2022-10-07:
Thank you, I built your environment with jumpers tonight, but when I tested, after the esp8266 is powered off, the rise and fall will return to before calibration after power on

qqxioud, 2022-10-07:
Hi, can you send me a sample code for esphome? I just started playing with this, thanks

RoadkillUK, 2022-10-07:
What do you mean by sample code. All the code is there.

Do you mean Automation examples?

qqxioud, 2022-10-08:
Hello, my ESPHome Current version: 2022.9.4, I copied them all to /config/esphome/, without editing and compiling testblind.yaml directly, the compilation error ERROR Error while reading config: Invalid YAML syntax:

Secret 'wifi' not defined
   in "/config/esphome/.common_blind.yaml", line 67, column 13:
         - ssid: !secret wifi
Do I need to edit the testblind.yaml file and how? Is there an example of this?
Still don't need to edit this file?
thanks

RoadkillUK, 2022-10-08:
Home assistant and ESPhome use a ```secrets.yaml``` file that holds information such as passwords etc.

Take a look at this [Youtube video](https://www.youtube.com/watch?v=eW4vKDeHh7Y) that explains ```secrets.yaml``` much better than I can.

then you need to take a look at ```.common_blind.yaml``` for _!secret_ and add them to the ```secrets.yaml```

I really hope that helps.