RoadkillUK/Motor-on-a-Roller-Blind-for-ESPHOME #9: Stepper vibrating

korvolle, 2023-04-15:
Hi, my stepper is vibrating when i try to open or close it, i have a wemos d1 mini and i wired it exactly as it says in the code, Any idea about what the problem can be? Also my 5v power source can supply more than enough current for the board and the motor so i dont think thats the issue.

RoadkillUK, 2023-04-16:
If the wiring is correct, what speed are you trying to run it at?

Have you had the stepper working using Arduino?

korvolle, 2023-04-16:
Thanks for answer. I solved it. I forgot to solder d2, d3 and d4 pins to the esp so only d1, 5v and gnd had contact with the esp. A very dumb misstake but now its working flawlessly!

RoadkillUK, 2023-04-16:
Then we shall close it ;)

closed 2023-04-16 by RoadkillUK