RoadkillUK/Motor-on-a-Roller-Blind-for-ESPHOME #14: Motor only turns one way

KobeNuyts, 2023-08-28:
Hello, my stepper motor only turns one way and not in the other.
It turns CW for up and for down, I can't get it to turn CCW.
I connected it like this D1 = IN1, D2 = IN3, D3 = IN2, D4 = IN4.
I already tried to swap D1 and D2 and also D3 and D4
I also tried another ULN2003 driver board.

But it still only turns CW.
Can someone pls help?

RoadkillUK, 2023-08-28:
I don't know, the wiring checks out, I've checked the original and it matches. [Link](https://www.thingiverse.com/thing:2392856)

KobeNuyts, 2023-08-28:
Maybe I need to change something on the code?

KobeNuyts, 2023-08-29:
I fixed it I made a stupid mistake and didn't switch the wires from D2 and D3, my bad.

RoadkillUK, 2024-01-10:
Derp, it happens :)

closed 2024-01-10 by RoadkillUK