RoadkillUK/Motor-on-a-Roller-Blind-for-ESPHOME #16: Cover with 1 stepper based on ESP32 controller hangs/freeze time often

vgomenyuk, 2024-03-18:
Hello. I'm trying to use your code based on ESP32 controller and I stepper with uln2003 driver.
Using exactly your code. The only difference is that I switched to ESP32 and changed the devicename to wrb-livingroom.
While I'm trying to move the cover up and down, stop it from time to time, move it to some different positions by using the cover's component widget in HA, or by pressing the physical button to OPEN/CLOSE the cover, the controller hangs, then waiting some time and reboots, or even can hang forever, then need to do restart it by power. During this time I see next in the logs:
```
WARNING wrb-livingroom @ 192.168.1.84: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for wrb-livingroom @ 192.168.1.84
WARNING Disconnected from API
INFO Successfully connected to wrb-livingroom @ 192.168.1.84 in 0.008s
INFO Successful handshake with wrb-livingroom @ 192.168.1.84 in 2.794s

OR

[00:04:33][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[00:04:33][V][json:058]: Size after shrink 120 bytes
[00:04:33][W][component:214]: Component api took a long time for an operation (0.05 s).
[00:04:33][W][component:215]: Components should block for at most 20-30ms.
WARNING wrb-livingroom @ 192.168.1.64: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for wrb-livingroom @ 192.168.1.64
WARNING Disconnected from API
INFO Successfully connected to wrb-livingroom @ 192.168.1.64 in 0.008s
INFO Successful handshake with wrb-livingroom @ 192.168.1.64 in 0.097s

OR

[14:09:18][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[14:09:18][V][json:058]: Size after shrink 120 bytes
[14:09:18][VV][api.service:086]: send_cover_state_response: CoverStateResponse {
  key: 3901603252
  legacy_state: LEGACY_COVER_STATE_OPEN
  position: 1
  tilt: 0
  current_operation: COVER_OPERATION_IDLE
}
[14:09:18][W][component:214]: Component api took a long time for an operation (0.23 s).
[14:09:18][W][component:215]: Components should block for at most 20-30ms.
WARNING wrb-livingroom @ 192.168.1.64: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for wrb-livingroom @ 192.168.1.64
WARNING Disconnected from API
```

I'm trying to figure out an issue for a long time, but no luck. Please help. Much appreciate!

RoadkillUK, 2024-03-18:
I don't know what to say, I can't see changing it to an ESP32 would make a difference.  I've never had any problems with my blinds and I've had them (4) up for a few years now.  Have you tried a different ESP or power supply?

vgomenyuk, 2024-03-18:
Yes, I tried 3 different ESP32 models. Also I tried 2 different power adapters. An issue stays the same.
So I'm out of ideas for now.

Google says it's may be memory-related. 

Also on ESP32 page it was said "Attach a serial console and watch the logs. Correlate the disconnect you see in HA with what is displayed on the serial console at the time. If you don’t get anything useful increase the logger: level to VERY_VERBOSE and try again." But I didn't figure out how to attach a serial console to ESPHome device. Maybe you know how to do it?

vgomenyuk, 2024-03-23:
Fixed an issue. The problem is with ESP32 board power supply internal logic.
I powered my ESP32 board through 5V pin. I tried to add into the board powering schema such elements as diodes, transistors and capasitors - nothing helped.
An issue was fixed when I switched the board powering (by the same power adapter) from 5V pin to USB port.

closed 2024-03-23 by vgomenyuk

RoadkillUK, 2024-03-23:
Yeah, I've not mentioned this (but I've now added it to the readme file) I run my blinds with a 9v power supply connected to the VIN pin.