RoadkillUK/Motor-on-a-Roller-Blind-for-ESPHOME #17: Blinds timing out

MnrHuman, 2024-09-11:
Hi, 

Thank you very much for your code, it works really well. I spent a month developing my own code and it worked, but when I stumbled on your code it just worked so much better. And all the add-ons like the uptime timer, wifi signal strength, IP address and easy configuration. Really works well. 

So I've installed a larger blind, but it keeps gaining length. About 5mm after every operation. So I think with the larger blind the motor is slipping a bit. I've tried another stepper motor but it still happens. So I changed my 3D design to increase the gear ratio from 1:6 down to 1:24. So the motor has more torque but then takes longer to open and close. After my installation the blind only went about 30%. My motor is turning the wrong way so I just started the calibration from the bottom going up. But after about 4 minutes the blinds stop. Is there a timer in the code that stops the proccess?

You'll see in the image below when the calibration is started at the bottom it only goes up that far:
[image]

Here is my updated gearing ratio from 1:6 to 1:24. So it is much stronger, but takes more time:
[image]

Thanks again, 
Gerrit

RoadkillUK, 2024-09-13:
Hi, I've updated the code to allow for longer blinds, would you be so kind as to let me know if it works for you?

MnrHuman, 2024-09-16:
Hi, I tried it last night and it works. Thanks a lot for the help. I compared your code with the previous version and the change is quite simple. Thanks again for the help!

RoadkillUK, 2024-09-16:
Glad it worked, I don't have blinds big enough to test ;)