RoadkillUK/Motor-on-a-Roller-Blind-for-ESPHOME #18: Blind gaining length

MnrHuman, 2024-09-19:
Hi, I've been running my blind with the reduced gear ratio for two days since Sunday and it is still gaining distance. The start and end stops is going down. With my previous design the gear ratio was 1:6 and it was gaining distance so I thought the stepper motor is slipping. Then I changed my design and reduced the gear ratio to 1:24 (so it is 4x stronger). The blind is about 1.6m wide, and 1m long, so it isn't super large. The 1m takes about 12 minutes to open and 12 minutes to close. So I set a timer to open then close the blind every 15 minutes. It hasn't been running the whole 2 days. I think it opened about 20-30 times and it gained about 10cm (on the open and closed setpoint).

So last night I used very a simple stepper motor .yaml example to see if I try other code if that will help. And in the past 14 hours it hasn't gained any distance. So I think the code might be the problem. I'm still troubleshooting. 

Is there maybe something you can think of that might cause this? Have you had similar issues when developing your code?

[image]

RoadkillUK, 2024-09-19:
I've not had any issues at all, but I don't run the blinds for that long.  You may be better using lower gear ratio and a stronger motor, or using a different type of cover component, such as [endstop ](https://www.esphome.io/components/cover/endstop)or [time based](https://www.esphome.io/components/cover/time_based).

MnrHuman, 2024-09-20:
Noted thanks. I thought I'd take a chance and ask. I'll troubleshoot a few more things.

closed 2024-09-20 by MnrHuman